Trel725/chavrprog #1: Can not get it work

antst, 2016-12-15:
Hey,

I am trying to use your code (on OSX though, but this should'n be an issue).

And I always get an answer:
Device reported its revision [4.03]
No answer, sorry

I checked pinout 10 times already, and programmer works with BIOS chips.

I have this kind of programmer 
http://cktplus.com/image/cache/catalog/Products/1221006/1221006-1-500x500.jpg

But it should not matter, after all it all the same 6 pins (5v,GND, MISO, MOSI,CS,CLK/SCK).

How can I debug this?

antst, 2016-12-15:
I checked and got the same result with Linux

Trel725, 2016-12-20:
Hello,
The program does very simple thing - it send sequence of byte to device and read it's answer. The very beginning of communication is sending 0xAC 0x53  which is command for entering to programming mode and should read 0x1E byte back which means that programming activated. Your output says that communication between computer and CH341 is OK, but there is no 0x1E answer from device. Usually it is because of  misconnection or power problems. Also, if your device are clocked at high frequency (8 MHz internal or more external) it is possible that AVR just can't start at 3.3 V, so it is necessary to use external power supply.

P.S. Проект честно говоря достаточно низкокачественный и был написан от безысходности в отсутвии нормального программатора.

closed 2017-05-10 by Trel725