Trel725/chavrprog #2: It's working, pretty good

JuncoJet, 2017-02-17:
My gamebunion is broken, lost the bootloader, now fix it myself.
[image]
[image]

On the cygwin it not work. But it on the Linux work well.
[image]
[image]

Trel725, 2017-02-19:
The problem in Windows is connected with some hardware-dependent functions which are available in Linux kernel but not in Windows. Unfortunately, I am very bad in Windows system, so I did my best by compiling it for cygwin. It is possible to get it running on Windows, but it need some weird magic with drivers, which I can't understand.

P.S. Thank you for your reply, JuncoJet. I am happy that my little project was helpful for someone else.

closed 2017-05-10 by Trel725

HilmarSt, 2019-10-20:
I know that is was 2 years ago, but I had the same issue which can be solved with Zadig by replacing the CH341driver which was installed with CH341PAR.exe with libusb.
Be aware that ASprogrammer (for writing SPI flash chips) does not work after this - libusb has to be uninstalled to re-enable AS Programmer.