Trel725/chavrprog #3: Not working

tiagooliveira95, 2017-06-13:
Hi,

Im trying to program the atiny85 and it's not working, everything is connected but i allways get the message "No answer, sorry"

I tried with the atmega328 same result

Using ubuntu 16.4.2

Trel725, 2017-06-14:
Hello,
I have just checked how it is compiling and running on Ubuntu 16.4.2, and everything is OK. The problem seems to be in hardware - check all the connectivity, AVR voltage supply. I will add image of connections.
Here is program output example:

> ./chavrprog -d mega328p -f
> Device reported its revision [4.03] 
> AVR answered! 
> Signature read as 1e 95 f  
> Signature is correct 
> Fuses: low high ext 
> ff de fd 

> lsb_release -a
> No LSB modules are available.
> Distributor ID:	Ubuntu
> Description:	Ubuntu 16.04.2 LTS
> Release:	16.04
> Codename:	xenial
>

tiagooliveira95, 2017-06-14:
Hi,
Thanks for your replay :)

I guess i just had bad luck, looks like the brand new chips that i bought don't work, i tried to program with Arduino As Isp with no luck

I tried with another 328 that i had, and i got a responce, however the signature failed

> AVR answered!
>Signature read as 1E 95 F
>Signature mismatch, expected 1E 95 14
>Exiting

I get the same signature as you, but the signeture check fails

Trel725, 2017-06-14:
Try to use flag -d mega328p

tiagooliveira95, 2017-06-14:
Hi,
Thanks for your replay :)

I guess i just had bad luck, looks like the 2 brand new chips that i bought yesterday don't work, i tried to program with Arduino As Isp with no luck

I tried with another 328 that i had, i got a responce, i i was able to program using chavrprog.

Thanks alot for your time :)

closed 2017-06-14 by tiagooliveira95