Trel725/chavrprog #6: Trying to add Atmega 328pb on Windows

Arthrimus, 2019-06-07:
I have added an entry for the 328pb in the config.c file with the correct ID but when I run the command to program the chip I get the error "No such chip" and a listing of all of the available chips from the config file. My entry for the 328pb doesn't show up in this list.

I am using chavprog.exe on windows.

Trel725, 2019-06-07:
Hello, could you provide your workflow? I am wondering if program was recompiled after addition of your device to the config.c?

Arthrimus, 2019-06-07:
I did not recompile the program after editing the config.c file. Could you please explain how to do that in windows? I didn't see anything about it in the documentation.

Trel725, 2019-06-07:
In general, it's not very easy task to do. You need to install Cygwin ([for example](http://www.ecs.shimane-u.ac.jp/~kyoshida/cygwin_install.htm)) and normally compile program with "make". Unfortunately, I am not experienced in development under Windows, so I am not very helpful. I've heard that modern versions of Windows contains a Linux subsystem, so you probably may consider to use it. In that case compilation should be as easy as run "make" in the folder with modified source.

Arthrimus, 2019-06-07:
Ok I attempted to run make in Cygwin. I get the following error. I did add Libusb1.0 in the cygwin installer. Do you have any suggestions?

$ make
gcc -std=gnu99 -Wall -O2  ch341a.c main.c chavrprog.c config.c  ihex_copy.c ihex_parse.c ihex_record.c -o chavrprog -lusb-1.0
ch341a.c:21:10: fatal error: libusb-1.0/libusb.h: No such file or directory
 #include <libusb-1.0/libusb.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:7: chavrprog] Error 1

Trel725, 2019-06-08:
Please, post the output of the folowing command:
`cygcheck --check-setup | grep -i libusb`

Arthrimus, 2019-06-09:
$ cygcheck --check-setup | grep -i libusb
libusb1.0                1.0.21-1            OK

Trel725, 2019-06-09:
You need to install the development version of libusb, libusb1.0-devel.

Arthrimus, 2019-06-09:
That worked, thanks so much for your help!

closed 2019-06-09 by Arthrimus