Trel725/chavrprog #9: Issues (and hints) using chavrprog with an ATtiny85

HilmarSt, 2019-10-20:
Thank you for this very useful tool!

The included chavrprog.exe runs on my Windows 10 (1903) PC without errors. A Cygwin installation is not necessary when just adding cygwin1.dll (64 Bit) and libusb-1.0.dll in the local directory - maybe you mention this in the readme.

The included chavrprog.exe does not work on my system, reading fuses works but reading back the flash produces garbage: < 10% of the bytes are correct. Therefore I´ve started debugging...

I do not know how to add libraries to Cygwin so I've copied libusb-1.0.lib and libusb.h from github.com/libusb and changed the makefile to ... -o chavrprog -L./ -llibusb-1.0
and changed #include <libusb-1.0/libusb.h> to #include "libusb.h"
A make -B runs without errors/warnings.

When starting my chavrprog.exe it exits with "Failed to detach kernel driver: 'Cannot allocate memory'".
Doing a "//goto close_handle;" after libusb_detach_kernel_driver() miracly works.

After having wrong bytes at the beginning of flash pages I had to change DELAY (only used for usleep() in write_page()) from 4500 to 18000 , maybe it should be added as a command line option.

After all this, there are still 2 wrong bytes (00 00) at the start of the flash. Adding debug printf's to usbTransfer() showed that the 2 bytes are written twice: The first time with zero and the second time with the correct value (what does not work because the flash isn't erased any more).
As the flash write code isn't understandable (at least for me) I've done an ugly patch to change the first 2 written zero bytes to 0xFF - and everything is working fine now!

Trel725, 2019-10-30:
Hello, thank you for the  detailed report. I will extend readme with information you provided. 
Delay value is taken from the AVR documentation, but I suppose that it may vary for different families (4500 is for MEGA), or simply between different series. It is defintely good idea to add a command line option, I will work on it.
May I ask you for the binary you built? I don't have access to any Windows machine nor the hardware to test. But I will anyway add it to the project if it worked for you.

HilmarSt, 2019-10-31:
Here's the Windows64 binary that is working with an ATtiny85. It's the only device I have (tested), so I renamed it to chavrprogTiny85.exe
[chavrprogTiny85.zip](https://github.com/Trel725/chavrprog/files/3793806/chavrprogTiny85.zip)

Another hint:
Please show the exact command line syntax for writing a fuse in the help output. I've bricked a Digispark board by writing the wrong fuse value: I wanted to write 0xE2 (226) and executed "chavrprog -l 226" which resulted in "writing 26" and switching to an external clock source (which is not fitted).

HilmarSt, 2019-11-01:
I forgot to mention:
_chavrprogTiny85.exe -dtiny85 **-a** file.hex_
does not work, please use
_chavrprogTiny85.exe -dtiny85 -e_
followed by
_chavrprogTiny85.exe -dtiny85 -w file.hex_

Trel725, 2019-11-01:
Thank you for the information. I will add it to the README.

Trel725, 2019-11-02:
I've added the -t option and your modified version. Thank you for the ideas. I am closing the issue, but feel free to reopen it if anything else is needed.

closed 2019-11-02 by Trel725