Trel725/chavrprog #10: what config for ATmega1284P?

JPT77, 2020-04-21:
Hi, 
what config do I need for the 1284P?
Or, do you have any docs on the values in the config.c file?
thanks

btw, for the 1284 I think RESET goes to pin 5 SS not CS?

JPT77, 2020-04-21:
is this the correct code? I derived it from avrdude:

`  {"mega1284p",
  {0x1E, 0x97, 0x06},
  256,
  512,
  4096},`

btw, the 1284 got more than 64K flash, this seems to be a problem with many programmers. Will it work with chavrprog?

JPT77, 2020-04-21:
Compiles with warning, but does not work.

```
$ ./chavrprog -d mega1284p -r f TESTFILE
Device reported its revision [4.03]
No answer, sorry
Exiting...
```
Since I simply do not have enough knowledge i give up. 
Maybe it helps someone else.

Trel725, 2020-04-21:
Hello, your configuration seems OK, but unfortunately chavrprog does not support chips with more than 64K of flash. Such devices have slightly other system of commands for flashing, so adding support require a fair amount of extra work. 
No answer is almost certainly a wiring problem. Anyway, thank you for the report. Feel free to re-open issue if you have more comments.

closed 2020-04-21 by Trel725