Trel725/chavrprog #13: Not working for me...

jtannenba, 2021-09-21:
On Windows 10:
I have this: https://www.amazon.com/gp/product/B07R8S8V4H/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1
and followed this:  https://ericdraken.com/avr-amtel-attiny-ch341a-isp-programmer/ and I get device reported 4.03 no answer.
I am using the code from Issue #9 and have double checked all my connections.  The clip is connected ok as a USBasp can detect it ok.  I also am providing 5v to the chip instead of the 3.3v.  Loaded the driver with Zadig.  Any hints?  This programmer can see dip eeproms ok although I have to change the driver to CH341PAR.   I believe the  whole problem is which windows driver to use for this device?

Thanks,
Joe

HilmarSt, 2021-09-22:
Which device you would like to program? Please only use the code from Issue #9 if you have problems to communicate with an ATtiny85 . The voltage (3.3V or 5V) should not matter at room temperature.
You've wrote "The clip is connected ok as a USBasp can detect it ok." - does this mean you have an USBasp programmer? If yes, for what reason do you need chavrprog.exe?
I've good experience with the libusbk driver from Zadig version 2.4.x or 2.5  attached to the programmer, but also libusb-win32 worked if I remember right.
In general, a photo from the test set-up is often a great help...

jtannenba, 2021-09-22:
I am getting into surface mount and mostly with attiny85's.  I wanted multiple ways of programming them.  My setup is as this:   https://ericdraken.com/avr-amtel-attiny-ch341a-isp-programmer/.  Tried libusbk and same result. Possible that spi not working even though i2c is?  Haven't any spi eeproms to try.

HilmarSt, 2021-09-23:
I think you should also try the original chavrprog-master\windows\chavrprog.exe , the maintainer (Andrii) provided an additional "-t" parameter to adapt the timings to slower devices like ATtiny85 - see the end of the discussion in Issue #9 .
The I2C and SPI buses are totally independent. Be sure to put your DIL-8 adapter on the correct position in the DIL-24 socket: 24XX is for I2C and 25XX is for SPI. And be sure not to rotate the adapter by 180 degrees - i've done this in error one time (luckily without consequences).
To ensure that your cabling is correct, measure with an ohmmeter the routing of the SPI (MISO, MOSI, CLK, CS) and power lines from the pin headers near the socket to the ATtiny85 pins.

jtannenba, 2021-09-23:
Yeah, did all that.  Wiring been check many times over.  I believe this adapter is not working in the SPI side.  The I2C side talks to eeproms just fine.  Sadly, I have no SPI eeproms to try.  Since I can use an USBasp or a USBtiny to program with, I'm going to close this.

Thank you

closed 2021-09-23 by jtannenba