Trel725/chavrprog #20: supporte for device 1a86:5523

hmollercl, 2024-06-30:
currently it only support device 1a86:5512 but I have 5523
```

$ lsusb
Bus 002 Device 009: ID 1a86:5523 QinHeng Electronics CH341 in serial mode, usb to serial port converter
```

try changing only in ch341a.h
```
//#define     CH341A_USB_PRODUCT     0x5512
#define     CH341A_USB_PRODUCT     0x5523
```

but when trying to upload:
```
Erasing...
Device reported its revision [4.03]
ch341SpiStream: Failed to read 4 bytes 'Argument list too long'
No answer, sorry
Exiting...
```
so I assume  further investigation needs to be done

HilmarSt, 2024-07-01:
The CH341A is configured in "serial mode". I think this should be "parallel mode"?
May be you change the jumper setting and try again...

hmollercl, 2024-07-25:
changing the jumper from 2-3(TTL) to 1-2 doesn't work either, now output is:
```
Erasing...
Couldn't open device [1a86:5523].
No answer, sorry
Exiting...
```
instead of   with jumper in 2-3 (TTL) the output is
```
Erasing...
Device reported its revision [4.03]
ch341SpiStream: Failed to read 4 bytes 'Argument list too long'
No answer, sorry
Exiting...
```